fledge-iot/fledge #152: A fresh install of Fledge using the archives repo for Ubuntu server 18.04 on RPi 4 4GB shows errors in system log

sudhirkshirsagar, 2020-08-15:
Aug 15 02:34:01 ubuntu Fledge Storage[17992]: ERROR: Failed to register configuration category: {"error": {"message": "[AttributeError] 'NoneType' object has no attribute 'register'"}}.

Aug 15 02:33:59 ubuntu Fledge Storage[17992]: ERROR: Failed to register configuration category: {"error": {"message": "[AttributeError] 'NoneType' object has no attribute 'register'"}}.

Aug 15 02:33:59 ubuntu Fledge Storage[17992]: ERROR: HTTP error while adding child categories: 404: No such Advanced category exist

MarkRiddoch, 2020-08-26:
Thanks for bringing this to our attention, we have raised a Jira ticket for this [https://scaledb.atlassian.net/browse/FOGL-4451](https://scaledb.atlassian.net/browse/FOGL-4451) and plan to have this fixed in the next 2 week sprint cycle.

muslui, 2023-06-13:
this is still an issue in 2.1.0 with ubuntu 18.04. Any idea to solve it?
[image]
Purge operation fails when this error is presented on startup.

TheVexy, 2024-05-17:
Still an issue in 2.4