fledge-iot/fledge #173: Issue in ingesting nested dictionaries in a Python South plugin

sudhirkshirsagar, 2020-09-25:
Sep 24 21:27:07 sudhir-Aspire-5749 Fledge mqn1[13257] WARNING: sending_process: sending_process_mqn1: unable to convert in memory data structure related to the readings data - error details |float() argument must be a string or a number, not 'dict'| - row |{'id': 637, 'asset_code': 'sample', 'reading': {'name': 'A', 'foo': {'nest': 5}}, 'user_ts': '2020-09-25 04:27:03.899686', 'ts': '2020-09-25 04:27:04.396'}|

sudhirkshirsagar, 2020-09-25:
The readings appear in the dashboard but with the nested dictionary as an object. So the ingest may be OK but the processing may be an issue.

sudhirkshirsagar, 2020-09-25:
I will try it with http north

sudhirkshirsagar, 2020-09-26:
I get the same error with the http_north plugin. I will open this there.

sudhirkshirsagar, 2020-09-27:
I was able to use the httpc successfully with the nested json. Looking at the error a bit more, it is definitely from Fledge python code and not the north plugins.
{'id': 637, 'asset_code': 'sample', 'reading': {'name': 'A', 'foo': {'nest': 5}}, 'user_ts': '2020-09-25 04:27:03.899686', 'ts': '2020-09-25 04:27:04.396'}
The north plugin don't have variables like id or asset_code.

MarkRiddoch, 2020-09-30:
Thanks for this, we do have an issue with some destinations that they can not support nested objects and require that we flatten the structure, however it looks like you have found a couple of places where we do not correctly handle nesting. I will get this added to the project task list and looked into.

sudhirkshirsagar, 2020-09-30:
Thanks Mark. I got around by hacking your GCP code. I like your coding style!

ashish-jabble, 2020-12-23:
@MarkRiddoch We may close this out as we have support the nested objects for python north tasks in develop branch via PR https://github.com/fledge-iot/fledge/pull/232

MarkRiddoch, 2021-01-26:
A fix for this will be included in the 1.9 release of Fledge that is currently under preparation.

MarkRiddoch, 2021-02-25:
1.9 is now released and packages are available.

closed 2021-02-25 by MarkRiddoch