fledge-iot/fledge #183: Notification service creation did not work. Throws errors in the log.

sudhirkshirsagar, 2020-10-08:
It was the first time that I created a notification service instance in a working Fledge on Ubuntu 18.04 on i5 machine with 4GB ram.
The log shows:
Oct 8 12:38:24 sudhir-Aspire-5749 Fledge notiserv1[18559]: ERROR: Can not access plugin directory /usr/local/fledge/python/fledge/plugins/notificationDelivery/: No such file or directory

Oct 8 12:38:24 sudhir-Aspire-5749 Fledge notiserv1[18559]: ERROR: Can not access plugin directory /usr/local/fledge/plugins/notificationDelivery/: No such file or directory

Oct 8 12:38:24 sudhir-Aspire-5749 Fledge notiserv1[18559]: ERROR: Can not access plugin directory /usr/local/fledge/python/fledge/plugins/notificationRule/: No such file or directory

Oct 8 12:38:24 sudhir-Aspire-5749 Fledge notiserv1[18559]: ERROR: Can not access plugin directory /usr/local/fledge/plugins/notificationRule/: No such file or directory

ashish-jabble, 2020-10-09:
@MarkRiddoch We have already a FOGL-3212 JIRA ticket to handle. (backlog)

sudhirkshirsagar, 2020-10-09:
After the error, I tried to create notification services. They get enabled but don't work.

MarkRiddoch, 2021-01-26:
Fixes for this are now in the develop branch and should be release in the next few weeks as part of the Fledge 1.9 release.

MarkRiddoch, 2021-02-25:
1.9 is now released with these fixes.

closed 2021-02-25 by MarkRiddoch