fledge-iot/fledge #257: Adding the possibility to send more information to Fledge from south plugin

Slayce, 2021-01-26:
It is currently not possible to attach as much data as we want to a particular datapoint.
 
For instance the iec-104 protocol allows to send quality descriptor of a specific measure, but this information can't be forwarded to Fledge apart from sending it in a different datapoint or reading, which isn't convenient.

aklira, 2021-01-27:
Hello,
I concur with this proposition. For each IEC 104 ASDU incoming message we need to be able to handle the following attributes: COT (cause of transmission), OA (originator address), CA (common address), isTest flag, isNegative flag. And for each IO (information object) within the ASDU message, we need to be able to handle the following attributes: IOA (information object address), Value and Quality Descriptor. 

I would suggest to enhance the current DataPoint with a parameters[], an array of JSON objects containing name/value pairs of the parameters to the incoming message. This would allow us then to handle every kind of incoming messages.

MarkRiddoch, 2021-02-25:
Looking at the C++ API's for datapoints and datapoint values we do support having arrays and lists in datapoints.

Array Support:
You may create an array of floats by creating a datapoint with a datapoint values that is created with a vector of doubles.
E.g.

```
vector<double> values;
values.push_back(1.0);
values.push_back(1.1);
DatapointValue dpv(values);
Datapoint("values", dpv);
```

Which gives an equivalent JSON output of

```
   "values" : [1.0, 1.1 ]
```
An array of name value pairs can also be created if you use a vector of datapoints

```
DatapointValue dpv1(1.0);
DatapointValue dpv2(1.1);
vector<Datapoint *> values;
values.push_pack(new Datapoint("first", dpv1));
values.push_back(new Datapoint("second", dpv2));
DatapointValue dpv(values, false);
```

Which gives an equivalent in JSON of

```
   "values" : [
         { "first" : 1.0 },
         { "second" : 1.1 }
    ]
```

Another option is to used nested objects

```
DatapointValue dpv1(1.0);
DatapointValue dpv2(1.1);
vector<Datapoint *> values;
values.push_pack(new Datapoint("first", dpv1));
values.push_back(new Datapoint("second", dpv2));
DatapointValue dpv(values, true);
```

Which gives an equivalent in JSON of

```
   "values" : {
         "first" : 1.0,
         "second" : 1.1
    }
```

closed 2021-03-13 by Slayce