fledge-iot/fledge #380: Missing Information how to start after installation

lukonjun, 2021-05-05:
Hi, 
I am interested in Edge Orchestrators and want to learn more about fledge. I managed to complete the installation:
`root@ubuntu-18:~/fledge# $FLEDGE_ROOT/bin/fledge status
Fledge v1.9.0 running.
Fledge Uptime:  69 seconds.
Fledge records: 0 read, 0 sent, 0 purged.
Fledge does not require authentication.
=== Fledge services:
fledge.services.core
fledge.services.storage --address=0.0.0.0 --port=42575
=== Fledge tasks:`
However I cant find instructions or documentation for the next step to deploy applications, could you please provide me some hints. 
Also is there an integration with Kubernetes possible?
Thanks and regards Lucas

shantanoo-desai, 2021-07-29:
I can do a write up on this to just help someone getting started with Fledge with CoAP + Sinusoid South Plugins. It will be up in a day or two.
About the Kubernetes Part, I am not sure I don't even know if there is an Docker/OCI container for Fledge and it's plugins out there.

shantanoo-desai, 2021-07-29:
Maybe this introduction might help https://shantanoo-desai.github.io/posts/technology/fledge_iot_rpi4/