fledge-iot/fledge #439: aptitude installation for Fledge v1.9.1 fails

shantanoo-desai, 2021-07-29:
## Hardware Spec
- Raspberry Pi 4 2GB Model B
- Ubuntu 20.04.2 LTS

## Steps to reproduce problem
1. Install GPG key for Fledge

        wget -q -O - http://archives.fledge-iot.org/KEY.gpg | sudo apt-key add -

2. Since there is no source for Ubuntu 20.04 LTS I use the 18.04

         sudo add-apt-repository ‘deb http://archives.fledge-iot.org/latest/ubuntu1804/aarch64/ / ‘

3. Update apt repositories

         sudo apt update

4. Install `fledge`

         sudo apt fledge

## Error Logs

The package configuration fails at this point:

```bash
cp cannot stat '/usr/local/fledge/data.new/extras/fogbench/fogbench_sensor_coap.template.json': No such file or directory
```

## Other Trials

1. I have tried purging the repository
 
         sudo apt remove --purge fledge

2. Removed `/usr/local/fledge` directory
3. reinstalling fledge and provides the same error

MarkRiddoch, 2021-07-30:
I have raised this in the engineering task system as FOGL-5769, it should be worked on shortly and a fix will be forthcoming., Thank you for raising this.

ashish-jabble, 2021-08-03:
Hi @shantanoo-desai
> Since there is no source for Ubuntu 20.04 LTS I use the 18.04

We officially do not support ub20.04 on `aarch64` architecture and with ub18.04 packages install on the given hardware specs which we do not recommend. There is no guarantee of packages to be work as these packages were made/compiled on ub18.04 platform specific.

We have updated the documentation in develop branch, you may see [here](https://fledge-iot.readthedocs.io/en/develop/quick_start/installing.html#ubuntu-or-debian)

> cp cannot stat '/usr/local/fledge/data.new/extras/fogbench/fogbench_sensor_coap.template.json': No such file or directory

This error comes only when fledge data directory exists in your system and above fogbench template is missing inside data directory which is required for the case.
Also you may see [doc](https://fledge-iot.readthedocs.io/en/develop/quick_start/uninstalling.html) for uninstallation of package

**Note** - sudo make install | sudo apt install fledge (as a service) - Both are stored in directory /usr/local/fledge by default.

I see you have tried purging of a repository & fledge installation directory removal. So I suspect there is something either a misconfigured/incomplete package installation or cache related issue which is causing it. Would you mind to clean your cache?
```
APTs cached files are located in:
/var/cache/apt/archives/

sudo apt-get clean
It removes everything except the partials folder and lock file from /var/cache/apt/archives/.

# if no success with clean command, try below
sudo dpkg --remove --force-remove-reinstreq fledge 
```

closed 2021-08-05 by ashish-jabble