fledge-iot/fledge #798: get_plugin_info tool SIGSEGV on bad arguments

jchabod, 2022-08-25:
In file `C/plugins/utils/get_plugin_info.cpp`, The application will SIGSEGV in ARGC=2,

Replace 
```
  if (argc<2)
```

by 
```
  if (argc<3)
```

MarkRiddoch, 2022-08-26:
Thanks for raising this. I will get this seen fixed.

jchabod, 2022-08-29:
Note:
As the prgram only work for "plugin_init" function, Isn't ist better to remove second argument and make it hard-coded?

Anyway,  this clearly is a very minor fix... 

Regards.

MarkRiddoch, 2022-09-21:
A fix for this is now in the develop branch. Also the function has been made an optional argument. This will be released in the next patch release, 2.0.1.

ashish-jabble, 2023-03-07:
Fixed in PR-https://github.com/fledge-iot/fledge/pull/833 and available onwards v2.0.1

closed 2023-03-07 by ashish-jabble