fledge-iot/fledge #843: Invalid Filter plugin interface

jchabod, 2022-10-03:
I'm using Fledge v2.0.0

There seem to be an inconsistency in FILTER plugin API:
- https://fledge-iot.readthedocs.io/en/develop/plugin_developers_guide/06_filter_plugins.html states:
```
 * @param handle     The plugin handle returned from plugin_init
...
void plugin_ingest(PLUGIN_HANDLE *handle,
                   READINGSET *readingSet)
{
        LogFilter *log = (LogFilter *) handle;
        log->ingest(readingSet);
}
```

But : `typedef void * PLUGIN_HANDLE;` is defined in `plugin_api.h` 
Thus `handle` is a `void**` and C-style casted to `LogFilter*`, which in no way can be a valid cast. (The comment `The plugin handle returned from plugin_init` is not helping here)

We may infer that either:
- The correct API is `void plugin_ingest(PLUGIN_HANDLE handle, READINGSET *readingSet`
- Or, the CONTENT of handle holds the pointer value : ` LogFilter *log = (LogFilter *) (*handle);`

jchabod, 2022-10-03:
Note: in `filter_plugin.cpp`, it seems that the correct API is `void plugin_ingest(PLUGIN_HANDLE handle, READINGSET *readingSet`

MarkRiddoch, 2022-10-05:
Thanks, I suspect this might be a documentation issue. We will need to look into it.

mbourhis, 2023-10-12:
It seems that issue concerns the *code* of at least one 'filter' plugin too.

These 2 'filter' plugins don't have the same API:

https://github.com/fledge-iot/fledge-filter-change/blob/develop/plugin.cpp
https://github.com/fledge-iot/fledge-filter-log/blob/develop/plugin.cpp

In one case, we can see
(https://github.com/fledge-iot/fledge-filter-change/blob/c82d07a4e520e9b8b4da68fcd5c83221ca718d28/plugin.cpp#L204)
```
void plugin_reconfigure(PLUGIN_HANDLE handle, const string& newConfig)
```
and
(https://github.com/fledge-iot/fledge-filter-change/blob/c82d07a4e520e9b8b4da68fcd5c83221ca718d28/plugin.cpp#L213)
```
void plugin_shutdown(PLUGIN_HANDLE handle)
```

and in the second case : 
(https://github.com/fledge-iot/fledge-filter-log/blob/6fe471d099137f1548f2cca56f43110ed7b767bf/plugin.cpp#L110)
```
void plugin_reconfigure(PLUGIN_HANDLE *handle, const std::string& newConfig)
```
and
(https://github.com/fledge-iot/fledge-filter-log/blob/6fe471d099137f1548f2cca56f43110ed7b767bf/plugin.cpp#L119)
```
void plugin_shutdown(PLUGIN_HANDLE *handle)
```