fledge-iot/fledge #993: North Plugin OMF

srjhunjhunwalacorp, 2023-03-01:
Hello Team,

I am trying to write data to Osisoft PI server using WebAPI , I have referred the documentation and set the configuration as referred and getting following error - 

FATAL: OMF Endpoint is not available

Can you please provide any pointer here to resolve this issue?

srjhunjhunwalacorp, 2023-03-08:
I am able to write it by changing the api address of the server to IP , looks like there was an issue with the DNS resolution, good to update this in the documentation.

closed 2023-03-08 by srjhunjhunwalacorp

MarkRiddoch, 2023-03-08:
Looking at the configuration you sent me the problem appears to be in the setup. The Server hostname has a URL configured when it should just be the hostname or an IP address of the host running the PI Web PI endpoint.