fledge-iot/fledge #1014: ConfigCategory constructor causes crash in south service

mzillgith, 2023-03-15:
The following south plugin code causes a crash of the south service when the configuration of the service is updated:

>     void plugin_reconfigure(PLUGIN_HANDLE *handle, string &newConfig)
>     {
>         ConfigCategory config("new", newConfig);
>     }

It seems the ConfigCategory constructor causes a stack corruption.

Fledge version: 2.1.0

Here are some messages from the log:
```
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: Signal 6 (Aborted) trapped:
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (0) 0       0x5650b64bda12 handler(int) + 82---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (1) 1       0x7f8a547b3090 /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f8a547b3090]---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (2) 2       0x7f8a547b300b gsignal + 203---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (3) 3       0x7f8a54792859 abort + 299---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (4) 4       0x7f8a547fd26e /lib/x86_64-linux-gnu/libc.so.6(+0x8d26e) [0x7f8a547fd26e]---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (5) 5       0x7f8a5489faba __fortify_fail + 42---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (6) 6       0x7f8a5489fa86 /lib/x86_64-linux-gnu/libc.so.6(+0x12fa86) [0x7f8a5489fa86]---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (7) 7       0x7f8a5205f43d plugin_reconfigure + 477---------
Mar 15 11:56:16 develop-VirtualBox Fledge s2opcua-1[66044]: FATAL: (8) 8       0x5650b6503d12 SouthPlugin::reconfigure(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) + 82---------
```

MarkRiddoch, 2023-03-15:
Would it be possible to attach an example of the configuration category?

mzillgith, 2023-03-16:
The problem can be reproduced with the default_config here: https://github.com/fledge-power/fledge-south-iec104/blob/1e3644eec04477bc35d6cc41f5838beafc7998d6/src/plugin.cpp#L31

But it also happend with the default configurations with the other south plugins I tested:

https://github.com/fledge-power/fledge-south-hnz
https://github.com/fledge-iot/fledge-south-s2opcua

MarkRiddoch, 2023-03-16:
Thank you for adding the information. I believe this is the same thing that the code that Is currently under review in the FOGL-7503 branch is aimed at fixing. Hopefully this will be in the develop branch very soon.

MarkRiddoch, 2023-03-22:
The fix for this is now in the develop branch and will be included in the next release.

mzillgith, 2023-03-22:
Great. I will have a look. Thank you

MarkRiddoch, 2023-08-04:
Have you had a chance to test this fix? If so can we close this issue?

mzillgith, 2023-10-16:
Seems to be fixed. Thanks.

closed 2023-10-16 by mzillgith