fledge-iot/fledge #1111: Use newer Ubuntu as base version

ReneWassinkAlliander, 2023-07-10:
The present Fledge build is based on Ubuntu 20.04, which is > 3 years old by now. It would be good to re-base Fledge on a newer Ubuntu release (for example: 22.04 LTS). This helps by providing a newer version of (for example) Python (version 3.10) and gives a longer-running support.

MarkRiddoch, 2023-07-10:
We are currently working on Ubuntu 22.04, the problem that is delaying this is that 22.04 comes by default with a new Python version and we have some compatibility issues with the various Python libraries we are using.

Ubuntu 22.04 will be made available as soon as we can resolve these issues.

Sander3003, 2023-07-10:
Example problem: the collection library is one of the problems

Upstream contributions are needed in Python libraries to fix this.

chicco785, 2024-02-28:
Is there any update on this matter? We would like to implement a south plugin for rabbit streams, but the related python library requires 3.9 at least.

geofjamg, 2024-03-19:
The only problem with Ubuntu 22.04 is the Python default version 3.10 but there is an easy way to install Python 3.8 with deadsnakes repository https://linuxgenie.net/how-to-install-python-3-8-on-ubuntu-22-04/.
The only isssue in that anywhere in the code this `python3` binary which is seached but `python3.8` is the one installed by deadsnake package.
Maybe just allowing to chose the python binary name would be a quick and easy way to support Ubuntu 22.04 ?

MarkRiddoch, 2025-01-24:
Ubuntu 22 packages are now available

ashish-jabble, 2025-09-22:
With the 3.1.0 release, Fledge now supports Ubuntu 22.04 LTS and Ubuntu 24.04 LTS.

closed 2025-09-22 by ashish-jabble