fledge-iot/fledge #1214: Fledge build errors with RaspiOS "Debian - Bookworm"

jpiwek, 2023-11-01:
Hi,

I am planning to create a Fledge IIoT Raspi CM4 Gateway  Demo system for Offgrid Power Supplies.
I removed python-dev packge from requirements.sh script.
However I get build errors when executing make:

`
pi@rpibw-dfe0:~/fledge $ make
Building Fledge version 2.1.0, DB schema 59
scripts/certificates "fledge" "365"
Creating a self signed SSL certificate ...
Certificates created successfully, and placed in data/etc/certs
scripts/auth_certificates ca "ca" "365"
Creating ca SSL certificate ...
ca certificate created successfully, and placed in data/etc/certs
scripts/auth_certificates user user "365"
Creating user SSL certificate ...
user certificate created successfully for user, and placed in data/etc/certs
scripts/auth_certificates user admin "365"
Creating user SSL certificate ...
user certificate created successfully for admin, and placed in data/etc/certs
cd cmake_build ; cmake /home/pi/fledge/
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

System is not RHEL/CentOS 7
CMake Deprecation Warning at C/common/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Error at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.11.2")
Call Stack (most recent call first):
  /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.25/Modules/FindPython/Support.cmake:3240 (find_package_handle_standard_args)
  /usr/share/cmake-3.25/Modules/FindPython3.cmake:490 (include)
  C/common/CMakeLists.txt:42 (find_package)

CMake Warning (dev) in /usr/share/cmake-3.25/Modules/FindPython/Support.cmake:
  Policy CMP0011 is not set: Included scripts do automatic cmake_policy PUSH
  and POP.  Run "cmake --help-policy CMP0011" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  The included script

    /usr/share/cmake-3.25/Modules/FindPython/Support.cmake

  affects policy settings.  CMake is implying the NO_POLICY_SCOPE option for
  compatibility, so the effects are applied to the including context.
Call Stack (most recent call first):
  /usr/share/cmake-3.25/Modules/FindPython3.cmake:490 (include)
  C/common/CMakeLists.txt:42 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/pi/fledge/cmake_build/CMakeFiles/CMakeOutput.log".
make: *** [Makefile:262: /home/pi/fledge//cmake_build/Makefile] Fehler 1
`

MarkRiddoch, 2023-11-14:
Thank you for raising this. We intend to add support for Bookworm in the near future

jpiwek, 2023-11-18:
@MarkRiddoch: Meanwhile I got at least Fledge + Fledge GUI running on RaspiOS Bookworm.
How can I get in contact with you? Would like to cover few things about a new industrial Reference Design under NDA.

thingstanje, 2024-09-03:
Hi, I dont know if i can share my experience building & running fledge on debian bookworm using default OS setup.
1. Add --break-system-packages  as below fin requirements.sh
[image]

3. Change the version of aiohttp, yarl, pyjq and ccharset in python/requirements.txt as picture below

[image]

Dibyajyoti-08, 2024-11-05:
Is there any update that fledge is now supporting for the debian 12 bookworm?