fledge-iot/fledge #1274: Rule/Delivery plugins remain loaded after disabling Notification instance

FlorentP42, 2024-01-22:
**Describe the bug**
When unchecking the "enable" box related the the whole Notification instance in the options, the rule plugin and delivery plugin used by said notification instance remain loaded. Which means that they can still perform internal tasks on their own when they should not be able to do so.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Notification Rule plugin which starts a thread that will prints "RULE ACTIVE" every second.
2. Create a Notification Delivry plugin which starts a thread that will prints "DELIVERY ACTIVE" every second.
3. Create a Notification Instance using the above rule and delivery plugins and enable it.
4. Notice that both "RULE ACTIVE" and "DELIVERY ACTIVE" logs are visble.
5. Disable the Notification instance from the instance options.
6. Notice that both "RULE ACTIVE" and "DELIVERY ACTIVE" logs are still visble.

**Expected behavior**
When a notification instance in disabled, its rule and notify plugins are unloaded and, as so, cannot perform any sort of action.

**Additional context**
Fledge 2.3.0

MarkRiddoch, 2024-01-23:
We have never unloaded plugins, if they have running threads then there is nothing we can do that will unload them. The plugin will be shutdown when the notification is disabled and this shutdown should stop any threads the plugin is running. I would suggest this is a bug int he plugin if the shutdown operation is not stopping the threads it started.

FlorentP42, 2024-01-23:
@MarkRiddoch Here is a sequence of logs of what happens:
```
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Configuration change notification for systemsp
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Notification systemsp disabled
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: DEBUG: Notification instance systemsp: removed subscription rulesystemsp for asset CONSTAT-1
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: DEBUG: Instance systemsp marked as Zombie
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loading rule plugin 'systemspr'.
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loaded rule plugin 'systemspr'.
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loading delivery plugin 'systemspn'.
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loaded delivery plugin 'systemspn'.
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspr - ConfigPlugin::importExchangedData : Connection loss tracking is active
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspr - ConfigPlugin::importAsset : Connection loss asset tracked: CONSTAT-1
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - ConfigPlugin::m_importDatapoint : Configuration prt.inf on GAMBS_0_T.GENE_ECLAIRAG_TS : [S_1340_0_0_33, SpsTyp]
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - ConfigPlugin::m_importDatapoint : Configuration access on GAMBS_0_T.GENE_PRESENC1_TS : [S_1340_0_0_16, SpsTyp, 10]
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::startCycles :  Starting configured cycles...
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::stopCycles :  Stopping all existing cycles...
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::stopCycles :  Cycles stopped!
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::startCycles :  Cycles started!
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: DeliveryPlugin::setEnabled = 1
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::runCycles :  Status Point cycle thread running for GAMBS_0_T.GENE_PRESENC1_TS
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::sendReading :  Creating and sending asset 'GAMBS_0_T.GENE_PRESENC1_TS' with reading { "PIVOT": { "GTIS": { "Identifier": "S_1340_0_0_16", "Cause": { "stVal": 3 }, "SpsTyp": { "stVal": 1, "q": { "Source": "substituted" }, "t": { "SecondSinceEpoch": 1706010458, "FractionOfSecond": 1124073 } }, "TmOrg": { "stVal": "substituted" } } } }
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: ERROR: systemspn - NotifySystemSp::ingest :  Callback is not defined
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: Adding AssetTracker tuple for service systemsp: systemspn:GAMBS_0_T.GENE_PRESENC1_TS:Notify, deprecated state is 0
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: Adding AssetTracker tuple for service systemsp: systemspn:GAMBS_0_T.GENE_ECLAIRAG_TS:Notify, deprecated state is 0
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: Zombie instance systemsp detected, deleting it ...
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: Calling plugin_shutdown for delivery deliverysystemsp
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::stopCycles :  Stopping all existing cycles...
Jan 23 12:47:39 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::runCycles :  Status Point cycle thread stopped for GAMBS_0_T.GENE_PRESENC1_TS
Jan 23 12:47:39 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::stopCycles :  Cycles stopped!
Jan 23 12:47:39 158319bca2e9 Fledge notif[323]: DEBUG: Skipping notification systemsp, notification is not enabled
Jan 23 12:47:39 158319bca2e9 Fledge notif[323]: INFO: Succesfully disabled notification instance 'systemsp'
Jan 23 12:47:44 158319bca2e9 Fledge notif[323]: DEBUG: Skipping notification systemsp, notification is not enabled
Jan 23 12:47:48 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::sendReading :  Creating and sending asset 'GAMBS_0_T.GENE_PRESENC1_TS' with reading { "PIVOT": { "GTIS": { "Identifier": "S_1340_0_0_16", "Cause": { "stVal": 3 }, "SpsTyp": { "stVal": 1, "q": { "Source": "substituted" }, "t": { "SecondSinceEpoch": 1706010468, "FractionOfSecond": 1124073 } }, "TmOrg": { "stVal": "substituted" } } } }
```
What seems to be happening is actually a bit different that what I initially understood: When I update the configuration of the Notification instance, the delivery and rule plugins semms to be loaded _again_, meaning there would be two instances of both running in parallel:
```
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loading rule plugin 'systemspr'.
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loaded rule plugin 'systemspr'.
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loading delivery plugin 'systemspn'.
Jan 23 12:47:37 158319bca2e9 Fledge notif[323]: INFO: Loaded delivery plugin 'systemspn'.
```
Then shutdown is indeed called on one of the delivery plugins (but not on the rule plugin ??):
```
Jan 23 12:47:38 158319bca2e9 Fledge notif[323]: DEBUG: Calling plugin_shutdown for delivery deliverysystemsp
```
But as we can see below there is still another delivery plugin running somewhere that was not shutdown as this log is being produces after the `plugin_shutdown()` was executed:
```
Jan 23 12:47:48 158319bca2e9 Fledge notif[323]: DEBUG: systemspn - NotifySystemSp::sendReading :  Creating and sending asset 'GAMBS_0_T.GENE_PRESENC1_TS' with reading { "PIVOT": { "GTIS": { "Identifier": "S_1340_0_0_16", "Cause": { "stVal": 3 }, "SpsTyp": { "stVal": 1, "q": { "Source": "substituted" }, "t": { "SecondSinceEpoch": 1706010468, "FractionOfSecond": 1124073 } }, "TmOrg": { "stVal": "substituted" } } } }
```

RVerhoeff, 2024-03-13:
Your assessment is correct: there was an instance of the Rule and Delivery support objects that remained in existence even when the notification was disabled. The problem is in the startup code that would create a notification instance in all cases, even when the notification was disabled. When the notification was later enabled, a second instance was created. This has been fixed. The startup code will not create a notification instance if it is disabled. There was never a memory leak because all instances were properly destructed when Fledge was shut down.

closed 2024-03-13 by RVerhoeff