fledge-iot/fledge #1315: Error at restart with a high number of services

geofjamg, 2024-03-20:
I have 150 south services and 10 north services and when I stop then start fledge I have the following error:
```
Mar 20 10:10:48 poc-fledge1 Fledge [1990997] INFO: script.fledge: Fledge started.
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.common.configuration_manager: Unable to create new category based on category_name iec104_s99 and category_description iec104_s99 an
d category_json_schema {'plugin': {'description': 'iec104 south plugin', 'type': 'string', 'readonly': 'true', 'default': 'iec104', 'value': 'iec104'}, 'asset': {'description': 'Asset name', 'type':
 'string', 'order': '1', 'displayName': 'Asset Name', 'mandatory': 'true', 'default': 'iec104', 'value': 'iec104'}, 'protocol_stack': {'description': 'protocol stack parameters', 'type': 'JSON', 'or
der': '2', 'displayName': 'Protocol stack parameters', 'default': '{ "protocol_stack" : { "name" : "iec104client", "version" : "1.0", "transport_layer" : { "redundancy_groups" : [ { "connections" : 
[ { "srv_ip" : "127.0.0.1", "port" : 2404 } ], "rg_name" : "red-group1", "tls" : false, "k_value" : 12, "w_value" : 8, "t0_timeout" : 10, "t1_timeout" : 15, "t2_timeout" : 10, "t3_timeout" : 20 } ] 
}, "application_layer" : { "orig_addr" : 10, "ca_asdu_size" : 2, "ioaddr_size" : 3, "asdu_size" : 0, "gi_time" : 60, "gi_cycle" : 60, "gi_all_ca" : false, "utc_time" : false, "cmd_wttag" : false, "c
md_parallel" : 0, "time_sync" : 0 } } }', 'value': '{ "protocol_stack" : { "name" : "iec104client", "version" : "1.0", "transport_layer" : { "redundancy_groups" : [ { "connections" : [ { "srv_ip" : 
"127.0.0.1", "port" : 2404 } ], "rg_name" : "red-group1", "tls" : false, "k_value" : 12, "w_value" : 8, "t0_timeout" : 10, "t1_timeout" : 15, "t2_timeout" : 10, "t3_timeout" : 20 } ] }, "application
_layer" : { "orig_addr" : 10, "ca_asdu_size" : 2, "ioaddr_size" : 3, "asdu_size" : 0, "gi_time" : 60, "gi_cycle" : 60, "gi_all_ca" : false, "utc_time" : false, "cmd_wttag" : false, "cmd_parallel" : 
0, "time_sync" : 0 } } }'}, 'exchanged_data': {'description': 'exchanged data list', 'type': 'JSON', 'order': '3', 'displayName': 'Exchanged data list', 'default': '{ "exchanged_data": { "name" : "i
ec104client", "version" : "1.0", "datapoints" : [ { "label":"TM-1", "protocols":[ { "name":"iec104", "address":"41025-4202832", "typeid":"M_ME_NA_1" } ] }, { "label":"TM-2", "protocols":[ { "name":"
iec104", "address":"41025-4202852", "typeid":"M_ME_NA_1" } ] }, { "label":"TS-1", "protocols":[ { "name":"iec104", "address":"41025-4206948", "typeid":"M_SP_TB_1" } ] } ] } }', 'value': '{ "exchange
d_data": { "name" : "iec104client", "version" : "1.0", "datapoints" : [ { "label":"TM-1", "protocols":[ { "name":"iec104", "address":"41025-4202832", "typeid":"M_ME_NA_1" } ] }, { "label":"TM-2", "p
rotocols":[ { "name":"iec104", "address":"41025-4202852", "typeid":"M_ME_NA_1" } ] }, { "label":"TS-1", "protocols":[ { "name":"iec104", "address":"41025-4206948", "typeid":"M_SP_TB_1" } ] } ] } }'}
, 'tls': {'description': 'tls parameters', 'type': 'JSON', 'order': '4', 'displayName': 'TLS parameters', 'default': '{ "tls_conf" : { "private_key" : "iec104_client.key", "own_cert" : "iec104_clien
t.cer", "ca_certs" : [ { "cert_file": "iec104_ca.cer" }, { "cert_file": "iec104_ca2.cer" } ], "remote_certs" : [ { "cert_file": "iec104_server.cer" } ] } }', 'value': '{ "tls_conf" : { "private_key"
 : "iec104_client.key", "own_cert" : "iec104_client.cer", "ca_certs" : [ { "cert_file": "iec104_ca.cer" }, { "cert_file": "iec104_ca2.cer" } ], "remote_certs" : [ { "cert_file": "iec104_server.cer" 
} ] } }'}}
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration: Failed to create category.
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration: Traceback (most recent call last):
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/sources/fledge/python/fledge/services/core/api/configuration.py"
, line 149, in create_category#012    await cf_mgr.create_category(category_name=category_name, category_description=category_desc,
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/sources/fledge/python/fledge/common/configuration_manager.py", l
ine 1280, in create_category#012    all_categories = await self._read_all_category_names()
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/sources/fledge/python/fledge/common/configuration_manager.py", l
ine 616, in _read_all_category_names#012    results = await self._storage.query_tbl_with_payload('configuration', payload)
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/sources/fledge/python/fledge/common/storage_client/storage_clien
t.py", line 308, in query_tbl_with_payload#012    jdoc = await resp.json()
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/.local/lib/python3.8/site-packages/aiohttp/client_reqrep.py", li
ne 1098, in json#012    await self.read()
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/.local/lib/python3.8/site-packages/aiohttp/client_reqrep.py", li
ne 1036, in read#012    self._body = await self.content.read()
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/.local/lib/python3.8/site-packages/aiohttp/streams.py", line 375
, in read#012    block = await self.readany()
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/.local/lib/python3.8/site-packages/aiohttp/streams.py", line 397
, in readany#012    await self._wait("readany")
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration:   File "/home/gj/.local/lib/python3.8/site-packages/aiohttp/streams.py", line 304
, in _wait#012    await waiter
Mar 20 10:11:31 machine1 Fledge[1991148] ERROR: logger: fledge.services.core.api.configuration: aiohttp.client_exceptions.ClientPayloadError: Response payload is not completed
Mar 20 10:11:31 machine1 Fledge iec104_s99[1994560]: ERROR: Failed to add configuration category: Response payload is not completed.
```

I also have sometimes this kind of crash:
```

MarkRiddoch, 2024-03-27:
I suspect this is all triggered by the services starting together and overloading the API of the core service. We have done the first state of adding the ability to order the start up, which will help with this. But we need to probably complete that work south that we can stagger the startup of the south services. Or maybe even have a mechanism that automatically rate limits the service startup when we start getting large numbers.

geofjamg, 2024-03-29:
I tried with just one south service configured and when I stop then restart fledge I get more than 300 hundreds http requests across all services just for them to restart (there is non data acquisition). I excluded ping http requests.
```Starting Fledge v2.3.0.
POST /fledge/service HTTP/1.1
GET /fledge/service/category/Storage HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
POST /fledge/service/category HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
GET /fledge/service/category HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
POST /fledge/service/category/Advanced/children HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
POST /fledge/interest HTTP/1.1
..POST /fledge/interest HTTP/1.1
.PUT /storage/reading/query HTTP/1.1
PUT /storage/table/streams/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
GET /storage/table/scheduled_processes HTTP/1.1
GET /storage/table/schedules HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
GET /fledge/service/ping HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
.GET /fledge/ping HTTP/1.1
POST /fledge/interest HTTP/1.1
POST /fledge/service/category?keep_original_items=true HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/statistics/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
POST /fledge/service/category/Storage/children HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
PUT /storage/table/category_children/query HTTP/1.1
[... 225 lines cut ...]
PUT /storage/table/category_children/query HTTP/1.1
PUT /storage/table/configuration/query HTTP/1.1
GET /fledge/service/category/iec104_s1Security HTTP/1.1
POST /fledge/interest HTTP/1.1
POST /storage/reading HTTP/1.1
PUT /storage/schema/fledge/table/statistics HTTP/1.1
POST /storage/table/tasks HTTP/1.1
GET /fledge/service?name=Fledge%20Storage HTTP/1.1
PUT /storage/schema/fledge/table/statistics/query HTTP/1.1
POST /storage/schema/fledge/table/statistics_history HTTP/1.1
PUT /storage/schema/fledge/table/statistics HTTP/1.1
PUT /storage/table/tasks HTTP/1.1
POST /storage/table/tasks HTTP/1.1
GET /fledge/service?name=Fledge%20Storage HTTP/1.1
PUT /storage/schema/fledge/table/statistics/query HTTP/1.1
POST /storage/schema/fledge/table/statistics_history HTTP/1.1
PUT /storage/schema/fledge/table/statistics HTTP/1.1
PUT /storage/table/tasks HTTP/1.1
POST /storage/table/tasks HTTP/1.1
GET /fledge/service?name=Fledge%20Storage HTTP/1.1
PUT /storage/schema/fledge/table/statistics/query HTTP/1.1
POST /storage/schema/fledge/table/statistics_history HTTP/1.1
PUT /storage/schema/fledge/table/statistics HTTP/1.1
PUT /storage/table/tasks HTTP/1.1
```

geofjamg, 2024-04-02:
@MarkRiddoch I found the issue. Simple-Web-Server has a default request timeout of 5s:
[image]
This is a little bit short when restarting 160 services because the storage server is flooded of request during a few seconds.
I tried to increase it and all my services finally restart without any error.
Would it be ok for you to be able to configure in fledge the the timeout_request and the timeout_content so that we can adapt it better to use cases?

geofjamg, 2024-04-02:
The hack I did to make it working:
[image]