fledge-iot/fledge #1339: Optimization of Fledge Data Ingest Performance using filter_ingest_callback

kwitekrac, 2024-04-16:
The main problem here is how to make the Fledge data intake work faster when using `filter_ingest_callback`.

I have two plugins in my test setup:

- The `south plugin` which gets data from RabbitMQ asynchronously (Python)
- The `filter plugin` which decodes data using portobuf (Python)

During testing, it became evident that the `filter_ingest_callback` function is responsible for data loss or significantly slows down the process with different settings. Here's how it was observed:

I sent 51 messages per second, each containing multiple measurements. The `south plugin` received these messages and passed them to the `filter plugin`, which decoded them, resulting in about 18 measurements per set. These measurements were stored locally. When all messages were decoded, the `filter_ingest_callback` function updated asset values.

In one 10-second test, 985 measurements were streamed, resulting in 9850 asset updates.

My system specs are: Ubuntu Linux, 8 CPUs, 8GB RAM.

Test results:
I used the `Received` count to track how many assets were updated.
Changing the `Maximum Buffered Readings` setting had a big impact on the results:

Basic configuration:

`Maximum Buffered Readings`=100: `Received`=267
`Maximum Buffered Readings`=50: `Received`=327
However, I couldn't check assets due to this error: Fledge Storage[209]: ERROR: SQLite3 storage plugin raising error: too many terms in compound SELECT

So, I switched the metadata database to `Postgres`, and the configuration that worked better was:

Storage configuration:
`Storage Plugin`: postgres
`Readings Plugin`: sqllitememory
`Threads`: 10
Plugin configuration:
`Maximum Buffered Readings`=10: 1,070 updates
`Maximum Buffered Readings`=1: 9,850 updates

But with a lower value for `Maximum Buffered Readings` the intake process slowed down. In the last test where all assets were updated, Fledge took 40 seconds to update all assets, so it won't work in real time.

So, the question is: How can I optimize Fledge to make it work in real time?

MarkRiddoch, 2024-04-24:
I suspect there are a number of things at play here. Can I ask some question about your setup?

Is this a build you have den yourself or are you using one of the release builds?
The "too many terms in compound SELECT" is usually the result of an incorrectly configured sqlite build. The default sqlite storage plugin uses multiple databases and tables to store readings, in order to reduce contention on the database locks. the number of these is uses is limited by the number of unions we can do in a query. Which is where this error comes from.

I suspect the choice of Python for the plugins is also part of the issue. We are having too convert the data to and from Python objects, which takes time. Also Python is very poor at asynchronous operations, as it is only possible to run a single thread at any one time even on multiple core hardware such as you have. The problem being the Python GIL.

I am intrigued by the split approach, what is the south plugin ingesting and passing align the pipeline? Is it not a well formed JSON reading?

kwitekrac, 2024-04-25:
Thank you Mark for your response. 

Regarding the choice of Python for our project, it primarily stemmed from the availability and maturity of RabbitMQ stream libraries. After evaluating options, we found that the Python-based [rstream](https://github.com/qweeze/rstream) library better suited our needs compared to the [RabbitMQ C client](https://github.com/GianfrancoGGL/rabbitmq-stream-c-client).

The library we chose runs on asyncio and needs `Python 3.9`, so we had to make a special version of Fledge for it, which runs on `Ubuntu 20.04`. Although there might be compatibility problems with newer Linux versions, our setup seems fine with `PostgreSQL` as storage. But, I'm unsure if this could affect how well our system performs.

As for the south plugin, it receives messages representing `protobuf` payloads of internally defined objects, which are then ingested into the pipeline. The filter's role is to decode the `protobuf`, sift through messages based on content, and convert received objects to `PIVOT` objects before JSON dumping. Subsequently, the next `filter_ingest_callback` is invoked.

An intriguing aspect is the handling of multiple channel values within each `asset`. Currently, I unpack these messages to create an `asset` for each channel. However, I recently encountered a different approach in last TSC where one `asset` (PMU) contains multiple channels. This alternative could significantly reduce the number of assets from 1k to 50. Nevertheless, I'm uncertain about how to implement this approach effectively, especially considering whether other plugins, particularly `IEC104 North`, support such configurations. Unfortunately, I couldn't find any guidance in the wiki regarding this matter.

Any insights or suggestions on this front would be greatly appreciated.

MarkRiddoch, 2024-04-26:
The 1K assets would explain why the default storage plugin is struggling, although I am surprised it did not simply make se of overflows tables. This may be a difference with SQLIte on your system. The Postgres, SQLiteLB or SQLiteMemory plugin should all be fine.

How are you passing the protobuf from the south service to the filter? Are you converting it to a Databuffer datapoint type?

kwitekrac, 2024-04-27:
I believe so, though I'm uncertain, since the `Databuffer datapoint` is utilized in C plugins. 
Received message, which is in bytes format, is converted into the dictionary format outlined in the [documentation](https://fledge-iot.readthedocs.io/en/latest/plugin_developers_guide/03_south_plugins.html).
```
data = {
            'asset': self.asset_name,
            'timestamp': utils.local_timestamp(),
            'readings': reads
}
async_ingest.ingest_callback(c_callback, c_ingest_ref, data)
```
My code:
```
encoded_data=base64.b64encode(msg) # Encode the protobuf data using Base64
to_store = { # dict is required in 'readings'
    'base64_encoded' : encoded_data,
}
data = {
    'asset': self.stream,
    'timestamp': utils.local_timestamp(),
    'readings': to_store
}
try:
    async_ingest.ingest_callback(c_callback, c_ingest_ref, data)
```

After reviewing the [documentation](https://fledge-iot.readthedocs.io/en/latest/plugin_developers_guide/035_CPP.html?highlight=Databuffer%20datapoint), it seems that the `Reading` class corresponds to my `data` dictionary, while the `to_store` structure mirrors the `Datapoint` class. However, these connections are based solely on my interpretation while bridging Python and C documentation.

MarkRiddoch, 2024-04-30:
Interesting, you are certainly in uncharted territory are with the base64 encoding of the protobufs in Python.  I suspect the bottleneck might be in the processing of the large Python string that results from the base64encoding. In order to pass it down the filter chain we have to convert this to a C++ structure, pass it to the filter, only for the filter to then discover it needs to convert it back into a Python string.  The south service has no idea of the filter implementation, and hence the conversion from and to Python is happening for each buffer. 

This is only a suspicion currently, we would need to do some profiling to identify if this is really the cause, but I am reasonably certain of it. Do you have any feel for the size of the base64 encoded protobuf?

kwitekrac, 2024-05-01:
I introduced `Base64` encoding solely to accommodate storing the Protobuf payload within Fledge. Initially, I encountered an error of 'unsupported characters' when attempting to ingest raw bytes. On average, the size of received messages is around 1kB. I'm wondering if there's a more efficient way to transfer data between the Python-based plugin and storage.

Another thought is to merge the code from the `RabbitMQ Streams` South plugin with a filter. This approach would eliminate the necessity for data conversion. However, it contradicts Fledge's concept of reusable plugins.

MarkRiddoch, 2024-05-02:
> I introduced `Base64` encoding solely to accommodate storing the Protobuf payload within Fledge. Initially, I encountered an error of 'unsupported characters' when attempting to ingest raw bytes. On [...]

The problem is that each element in the chain knows nothing about the others, so we use a common format between the elements. We might be able to do something to keep multiple encoding of the protobuf to prevent the need to convert back and forth if nothing has change since the last Python component used it.

I would need to think a bit about this and it would take a reasonable effort to implement.
> 
> Another thought is to merge the code from the `RabbitMQ Streams` South plugin with a filter. This approach would eliminate the necessity for data conversion. However, it contradicts Fledge's concept of reusable plugins.

If it can be done easily it would be an interest test to confirm the hypothesis as to where the bottle neck is.

kwitekrac, 2024-05-10:
I consolidated the filter plugin's functionality into the south plugin, eliminating the need for inter-plugin data communication. The results were remarkable: Fledge exhibited a 29-fold increase in asset updates per second compared to the previous setup. This substantial improvement prompts me to question the viability of Python-based filter plugins.

MarkRiddoch, 2024-05-22:
Thank you for this. This confirms by thoughts that the issue is related to the conversion to and from Python objects. I have got a thought about how to improve this. One of the principles of the system is that each filter element knows nothing about the next or previous element, so we always end up converting from Python objects to C++ and in you case back again. We can probably enhance the pipeline management code to cache the Python objects and reuse them rather than do the conversion again. We may even be able to do late conversation and avoid both conversions. I will need to put some thought into exactly how best to do that and integrate it with the new complex pipeline features we are adding.

closed 2024-06-05 by kwitekrac