fledge-iot/fledge #1382: Sparkplug B Plugin broken in current Docker image?

bbking78, 2024-05-31:
Hi,

I just installed the current Docker image 54.204.128.201:5000/fledge:latest-ubuntu2004 like mentioned here: https://fledge-iot.readthedocs.io/en/latest/quick_start/installing.html#using-docker-containerizer-to-install-fledge

When I activate the Sparkplug B plugin I get this error in the System Logs:
May 31 20:06:03 83c091dcd8ef Fledge sparkplugb[261482] ERROR: logger: mqtt_sparkplug: Traceback (most recent call last):

May 31 20:06:00 83c091dcd8ef Fledge sparkplugb[261482] ERROR: logger: mqtt_sparkplug: NameError: name 'sparkplug_b_pb2' is not defined

May 31 20:06:00 83c091dcd8ef Fledge sparkplugb[261482] ERROR: logger: mqtt_sparkplug: File "/usr/local/fledge/python/fledge/plugins/south/mqtt_sparkplug/mqtt_sparkplug.py", line 238, in on_message#012 inbound_payload = sparkplug_b_pb2.Payload() 

Is it indeed broken or do I miss something? 

Thanks in advance.

ashish-jabble, 2024-09-26:
> When I activate the Sparkplug B plugin

You mean here to install the package via APT mechanism or try to add Southbound service?

ashish-jabble, 2024-12-10:
There were some syntax issues during protobuf parsing, but these have now been resolved and the fix is available in the **v2.6.0**.

closed 2024-12-10 by ashish-jabble