fledge-iot/fledge #1407: plugin_init is called when creating a disabled north/south service

FlorentP42, 2024-07-01:
**Describe the bug**
When creating a north/south service, you can select if the plugin behind that service will be enabled by default or not from the GUI (or from the API).
If you create a service **disabled** at creation, then `plugin_init` is called for that plugin.
And even more strange the plugin initialized this way then seems to "disappear" without even a call to `plugin_shutdown`, which may result in memory leaks.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new fledge-south-hnz service from Fledge GUI with log level set to "debug" at startup (through START_LOG_LEVEL).
2. Uncheck the "Enable" box in the last menu of service creation.
3. Notice that `plugin_init` is called in the logs of that plugin.
4. Enable the plugin in the options.
5. Notice that `plugin_init` is called again in the logs of that plugin (as expected).
6. Notice that but `plugin_shutdown` was not called on that plugin between the two `plugin_init`.

**Expected behavior**
If we make the service disabled at creation, we expect the service configuration to be saved for when it will be enabled later, but no plugin to be created at all.

**Additional context**
 - Fledge Version: 2.4.0

MarkRiddoch, 2024-11-20:
This is part of the dryrun option that was added in a release of Fledge some time ago. It shoudl however be callign plugin shutdown. We will raise a ticket to have this looked into. The process itself will be shutdown, so memory will be recovered, other resources may be more of an issue.

FlorentP42, 2024-11-21:
@MarkRiddoch Thanks for those precisions. So does this mean there is a configuration we can send to Fledge that would disable this "dryrun" option? Does it affect all Fledge plugins at once or is it per-plugin?

MarkRiddoch, 2025-01-24:
No, there is no way to disable the dry run. It calls plugin_init to get the default configuration to enable you to then edit it as part of the creation of plugin. Without this you would not be able to edit configuration of a plugin that was created disabled. The plugin_init call should not do anything destructive, it is there to return data and do basic initialisation only.

FlorentP42, 2025-01-24:
That makes sense, what confused us here is that we are generating audit message once new configuration is loaded, and as `plugin_init` also load a new configuration the audits were also generated from there with the default configuration, and then sent again when `plugin_init` was called after enabling the plugin with our custom configuration.
A way to know that `plugin_init` was called in the context of a dry run could maybe be handy in that case, o that we can discriminate between both uses and make sure we only send the audit when it is relevant for us.