fledge-iot/fledge #1464: Running Fledge in Raspberry Pi

Dibyajyoti-08, 2024-09-23:
So i followed the step for the installation, 
-> sudo ./requirements.sh
-> which gave me the following error,
`Package python-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-dev-is-python3

E: Package 'python-dev' has no installation candidate
`
-> I ignored as it says python-dev-is-python3 is replaced, so i went on to next command make, which gave me the following error,
`cd python ; python3 setup.py build -b ../python_build_dir ; cd /root/git/fledge/ ; cp python/requirements.txt python_build_dir/lib/.
running build
running build_py
cp -r python_build_dir/lib/* /usr/local/fledge/python
python3 -m pip install -Ir python/requirements.txt --no-cache-dir
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
make: *** [Makefile:327: python_requirements] Error 1
`
-> I cannot figure out what is the issue? Help from someone will be much appreciated.

Regards,
DJ

praveen-garg, 2024-09-23:
https://github.com/fledge-iot/fledge/issues/1214#issuecomment-2325517625 may help

Dibyajyoti-08, 2024-09-24:
Thanks for the help.

closed 2024-09-24 by Dibyajyoti-08