fledge-iot/fledge #1466: Unable to connect fledge backend

Dibyajyoti-08, 2024-09-25:
-> I cloned fledge gui into my ubuntu system and followed the follwing command,

`    $ curl -sL https://deb.nodesource.com/setup_16.x | sudo -E bash -
    $ sudo apt-get install nodejs`

`./build --clean-start`

-> After that i add event in nginx.conf as without that nginx was not getting started,

`events {
    worker_connections 1024;
}

http {
    # ubuntu
    include /etc/nginx/mime.types;

    # mac
    #include /usr/local/etc/nginx/mime.types;

    # windows
    # include C:/nginx/conf/mime.types;

    server {
        listen 8081;
        # windows, Paths in a configuration file must be specified in UNIX-style using forward slashes.
        # root <ABSOLUTE PATH TO dist directory content>;
        root dist/;

        server_name _;

        # Enable static gzip
        gzip_static on;

        # Disable dynamic compression (optional, and not recommended if you're proxying)
        gzip off;
        gzip_types text/css text/javascript application/javascript application/x-javascript;

        sendfile on;

        location / {
            index fledge.html;
        }
    }

}`

-> Then when i went on browser and opened gui as http://<my-system-ip>:8081, it redirected me to the URL - http://192.168.11.179:8081/#/setting?id=1

-> Also it shows DOWN in the GUI, any help will be appreciated.

Regards,
DJ

ashish-jabble, 2024-09-26:
@Dibyajyoti-08 
a) Is Fledge running? You can verify by API endpoint `http://IP:PORT/fledge/ping`
b) Do you install Fledge and Fledge-GUI on same machine?
c) If both are running on different machines, then make sure it should be on the same network

Dibyajyoti-08, 2024-09-26:
Hi @ashish-jabble,
Thanks for the reply.
->Is there connection to run fledge and fledge-gui to run on a same system?
-> I only cloned fledge-gui and followed the step, after the command $ yarn install && yarn start it indicates me to open browser with http://localhost:4200, when i open it shows,
[image]
-> it also shows the same when i go to the browser and type http://localhost:4200/fledge/ping, it automatically redirects me to the URL - http://localhost:4200/#/setting?id=1

ashish-jabble, 2024-09-26:
> ->Is there connection to run fledge and fledge-gui to run on a same system?

Not as such

> http://localhost:4200

GUI is running as Development Server by default port with 4200

> http://localhost:4200/fledge/ping

Instead of 4200 you should use 8081 as Fledge Core server is running by default on 8081 i.e http://localhost:8081/fledge/ping

As per your screenshot - In Connection setup - You have set `http | localhost | 8081`

I suspect Fledge is Not running on localhost. Please check the Fledge setup first and configure the correct details in GUI client. Thereafter you shall be able to communicate between Fledge and its GUI client.

closed 2024-09-26 by Dibyajyoti-08