fledge-iot/fledge #1481: [fledge-gui] Unable to login in HTTPS with certificate

FlorentP42, 2024-10-29:
_There does not seem to be an issue panel in the [fledge-gui](https://github.com/fledge-iot/fledge-gui) repository, is this intended?_

**Describe the bug**
Currently it is not possible to login from the GUI with a Fledge user using certificate over HTTPS as no login screen is ever displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Fledge in HTTP with no authentication.
2. Configure `/fledge/category/rest_api` with the following settings : `{"authentication":"mandatory","authMethod":"password"}`
3. Restart Fledge.
4. Login as "admin" to access to the Fledge user creation menus.
5. Create a Fledge user named "fledge" and configure it to login wtith certificate only.
6. Create a self-signed CA certificate and a "fledge" user certificate using this CA.
7. Add both certificates to the Fledge store.
9. Configure `/fledge/category/rest_api` with the following settings : `{"enableHttp":"false","authMethod":"certificate"}`
10. Restart Fledge.
11. Refresh the Fledge GUI page which goes back to "Connection Setup" screen.
12. Try to access the Fledge instance in HTTPS on port 1995.
13. Notice that an information banner appear for a split second, the the GUI reloads and remain on the same screen with no further details.

**Expected behavior**
The GUI displays a login screen where you can enter your Fledge username and upload the certificate file to perform the login request.

**Screenshots**
This is the information message displayed for a split second before the page reloads itself :
[image]

**Desktop (please complete the following information):**
 - OS : Windows 10 host (both Fledge and Fledge GUI running in Oracle Linux 8.5 dockers)
 - Browser : Firefox (also tested on Edge with same behavior)
 - Version : Fledge 2.5.0 / Fledge-GUI 2.5.0

**Additional context**
N/A

mshariq-nerd, 2024-10-30:
@FlorentP42,
Please confirm if the information banner visible to you on settings page and you have clicked on the <link> inside the banner message and accepted the self signed certificate?

FlorentP42, 2024-10-31:
> @FlorentP42, Please confirm if the information banner visible to you on settings page and you have clicked on the link inside the banner message and accepted the self signed certificate? ![link](https://private-user-images.githubusercontent.com/16384273/381481194-875d91ea-9775-4a70-abbb-a5d7e8e4613f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzNjI3NzgsIm5iZiI6MTczMDM2MjQ3OCwicGF0aCI6Ii8xNjM4NDI3My8zODE0ODExOTQtODc1ZDkxZWEtOTc3NS00YTcwLWFiYmItYTVkN2U4ZTQ2MTNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMxVDA4MTQzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTllMGZjY2U0YzExN2ZmODY5OWNlZGZiMmQ5NzZmODIyNDdlNjAwYjdmMGEyZWJjMjNmNzE4NTExMjJhMjliMGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.3_R3r69V3CZmzlzjZOoQZYnKgm94b8wMwzqzc5z54lQ)

I can't possibly click it, it appears for like one 10th of a second then the page reloads itself, I was only able to screenshot it because I launched the screen capture tool at the exact right time.

mshariq-nerd, 2024-11-05:
@FlorentP42,
Is there any error in browser dev console? If yes please provide that, better if you can provide a screen recording.
Also confirm port 1995 is accessible on host machine & mapped/exposed for container?

FlorentP42, 2024-11-05:
Well, I am very confused... Starting today I can no longer reproduce the issue, and the only event that changed since yesterday is a restart of my computer...

The containers running Fledge and Fledge-GUI are still the same (v2.5.0), and was I was experiencing the issue on more than one browser I assume a browser cache issue was not the cause either.

As far as I can tell when I was experiencing the issue the port 1995 was accessible from another machine on our network through curl requests and let us login as intended when provided the required certificate, so the Fledge instance was there and listening on the correct port.

Anyway as I can't reproduce it for now in order to provide more precise debug information, I guess I will close the ticket, and reopen it if I ever encounter this situation again.

Also now that the banner no longer disappear immediately, I noticed that it simply pointed to the page `https://hostname:1995/fledge/ping`, and once I got there I was indeed able to "accept the risk and connect anyway", which fixed the rest of Fledge GUI after a refresh.