fledge-iot/fledge #1494: User role documentation is inconsistent

FlorentP42, 2024-11-06:
By browsing the documentation I found two pages talking about user roles, and they do not match :  

- First one states that there is 5 roles, but does not detail their ID : https://fledge-iot.readthedocs.io/en/latest/securing_fledge.html#user-management
[image]

- Second one lists all the role ID and only give 2 of them : https://fledge-iot.readthedocs.io/en/latest/rest_api_guide/02_RESTauthentication.html?highlight=role#add-user
[image]

I believe those two elements represent the same information correct? So they should match in their definition.

MarkRiddoch, 2024-11-21:
Thanks for this, Jira 9279 raised to resolve this

MarkRiddoch, 2025-01-16:
The changes have now been merged into the develop branch

closed 2025-01-16 by MarkRiddoch