fledge-iot/fledge #1497: Missing Plugins

Dibyajyoti-08, 2024-11-18:
Hello folks, 
can anyone help me in this issue

-> I cloned fledge backend and build in one system(System1), then i cloned fledge-gui on ubuntu and build it as per the README, and then copied the dist directory and nginx.conf to the System1 and then i started the nginx with the fledge nginx.conf, then i went to the browser and tried to add plugins, at that point of time it shows as no plugins are there to install. I have tried the same procedure in one system where i have cloned and build fledge and fledge-gui, at that point of time when i add plugins, it will ask to install, and then it will show all the plugins options to install them individually, but in this current scenario it does not show any plugins that i can install.
Any body know about this issue,
Any possible help from you all will be appreciate.
Note - in North microservice its only show OMF as an option and in the South microservice it does not show any microprocess.

Regards,
DJ

Dibyajyoti-08, 2024-11-19:
I tried to install the archives but as my OS is debian 12 bookworm its not installing, is there any way i can make this work around for debian 12 bookworm?

Regards

ashish-jabble, 2024-11-19:
> I tried to install the archives but as my OS is debian 12 bookworm its not installing, is there any way i can make this work around for debian 12 bookworm?

@Dibyajyoti-08 We currently do not have support for Bookworm. Please refer to the list of [supported platforms](https://fledge-iot.readthedocs.io/en/latest/quick_start/platforms.html).

Dibyajyoti-08, 2024-11-19:
@ashish-jabble I am aware of the bookworm support is not available currently for the fledge-iot, i was just wondering whether we can make a way around to install full-feldged fledge-iot in the bookworm OS, i was able to install the developement mode installation, but adding plugins was the trouble,
Can you suggest any other way of adding plugins instead of going through archives?

Regards

Dibyajyoti-08, 2024-11-19:
i tried to download the .deb of aarch64 package file and followed the installation process

`sudo apt -y install fledge_2.6.0_aarch64.deb`

Error: 
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'fledge' instead of './fledge_2.6.0_aarch64.deb'
fledge is already the newest version (2.6.0).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up fledge (2.6.0) ...
Install python dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
dpkg: error processing package fledge (--configure):
 installed fledge package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of fledge-south-sinusoid:
 fledge-south-sinusoid depends on fledge (>= 2.6); however:
  Package fledge is not configured yet.

dpkg: error processing package fledge-south-sinusoid (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 fledge
 fledge-south-sinusoid
E: Sub-process /usr/bin/dpkg returned an error code (1)
`
I extracted the package and add --break-system-packages in the postinst script for the python dependencies but still the error is same 

I have also set a venv for the installation as the same issue i was facing when i was running requirements.sh for the dependencies, and the way i resolved is to remove python-dev from the script.

ashish-jabble, 2024-11-19:
> i tried to download the .deb of aarch64 package file and followed the installation process

@Dibyajyoti-08 If you try downloading from http://archives.fledge-iot.org, it likely won’t work, as we do not support Bookworm and its packages are unavailable. However, if you still want to install Fledge and its plugins, you can attempt a development-mode setup by following the 'make' based installation process with the suggested tweaks from #1214. Once that’s done, you can try installing the plugin in development mode as described in the [initial comment ](https://github.com/fledge-iot/fledge/issues/1497#issuecomment-2483118919)of the conversation.

Dibyajyoti-08, 2024-11-19:
@ashish-jabble  I have already installed fledge in the developement-mode, i just need to install the plugins, are you refering this below link to follow for the plugins installation?
https://fledge-iot.readthedocs.io/en/latest/building_fledge/06_testing.html#fogbench-a-brief-intro

ashish-jabble, 2024-11-19:
> i just need to install the plugins, are you refering this below link to follow for the plugins installation?
https://fledge-iot.readthedocs.io/en/latest/building_fledge/06_testing.html#fogbench-a-brief-intro

yupe, this is an example of Python based CoAP plugin. Replace your plugin with the steps mentioned

Dibyajyoti-08, 2024-12-05:
Hi Ashish,
I want to know where can i find C or python based application/configuration which are written for the plugins, are those accessible for the contributors?

closed 2024-12-13 by Dibyajyoti-08