fledge-iot/fledge #1515: Custom Plugin

Dibyajyoti-08, 2024-12-13:
I am trying to add a custom plugin for the south service, for which i created a cpp file of the plugin as well as plugin.cpp by taking the reference of **usb4704** plugin, i have compiled and generated the shared object file with the **make install** command i was able to place the shared object file in the **/usr/local/fledge/lib** directory, after which i copied the file to the /usr/local/fledge/plugin/south/myplugin directory, then i tried to register the plugin with the rest API command, 
```bash
curl -sX POST http://localhost:8081/fledge/service -d '{"name": "myplugin", "type": "south", "plugin": "myplugin", "enabled": true}'
```
and the console output that i got is,
{"message": "No such myplugin category exist"}

Can anyone help me what am i missing here?

ashish-jabble, 2024-12-13:
Would you mind to share the output of `ls -la /usr/local/fledge/plugin/south/myplugin`?

Dibyajyoti-08, 2024-12-13:
```bash
ls -la /usr/local/fledge/plugins/south/zigbee_uart/
total 72
drwxr-xr-x 2 root root  4096 Dec 12 12:51 .
drwxr-xr-x 5 root root  4096 Dec 11 12:12 ..
-rwxrwxr-x 1 root root 29176 Dec 12 12:51 libzigbee_uart.so
-rw-r--r-- 1 root root 29176 Dec 12 12:51 libzigbee_uart.so.1
```

ashish-jabble, 2024-12-13:
Above Curl Request is fine, Could you please share more details regarding the plugin info?
```
curl -sX GET "http://localhost:8081/fledge/plugins/installed?type=south&config=true"
```

Dibyajyoti-08, 2024-12-13:
sure,
$ ./get_plugin_info /usr/local/fledge/plugins/south/zigbee_uart/libzigbee_uart.so plugin_info | jq
{
  "name": "zigbee_uart",
  "version": "2.6.0",
  "type": "south",
  "interface": "1.0.0",
  "flag": 0,
  "config": {
    "plugin": {
      "description": "Zigbee UART Data Communication Module",
      "type": "string",
      "default": "zigbee_uart",
      "readonly": "true"
    },
    "asset": {
      "description": "Asset name to use for readings",
      "type": "string",
      "default": "zigbee_uart",
      "order": "1",
      "displayName": "Asset Name",
      "mandatory": "true"
    },
    "port": {
      "description": "UART port for Zigbee communication",
      "type": "string",
      "default": "/dev/ttyUSB0",
      "order": "2",
      "displayName": "UART Port",
      "mandatory": "true"
    },
    "baudrate": {
      "description": "Baud rate for UART communication",
      "type": "integer",
      "default": 9600,
      "order": "3",
      "displayName": "Baud Rate",
      "mandatory": "true"
    }
  }
}

Dibyajyoti-08, 2024-12-13:
It shows in the **http://localhost:8081/fledge/plugins/installed**
```bash
{
  "plugins": [
    {
      "name": "OMF",
      "type": "north",
      "description": "PI Server North C Plugin",
      "version": "2.6.0",
      "installedDirectory": "north/OMF",
      "packageName": ""
    },
    {
      "name": "sinusoid",
      "type": "south",
      "description": "Sinusoid Poll Plugin which implements sine wave with data points",
      "version": "2.6.0",
      "installedDirectory": "south/sinusoid",
      "packageName": "fledge-south-sinusoid"
    },
    {
      "name": "http_south",
      "type": "south",
      "description": "HTTP Listener South Plugin",
      "version": "2.6.0",
      "installedDirectory": "south/http_south",
      "packageName": "fledge-south-http-south"
    },
    {
      "name": "Random",
      "type": "south",
      "description": "Random data generation plugin",
      "version": "2.6.0",
      "installedDirectory": "south/Random",
      "packageName": "fledge-south-random"
    },
    {
      "name": "usb4704",
      "type": "south",
      "description": "Advantech USB-4704 Data Acquisition Module",
      "version": "2.6.0",
      "installedDirectory": "south/usb4704",
      "packageName": "fledge-south-usb4704"
    },
    {
      "name": "zigbee_uart",
      "type": "south",
      "description": "Zigbee UART Data Communication Module",
      "version": "2.6.0",
      "installedDirectory": "south/zigbee_uart",
      "packageName": "fledge-south-zigbee-uart"
    }
  ]
}
```
-> But its not getting register as it says **no such category exist**

Dibyajyoti-08, 2024-12-17:
Is it checking the category present inside the fledge.services.south?

ashish-jabble, 2024-12-17:
> Is it checking the category present inside the fledge.services.south?

Not really!
When you create the service, either via the GUI client or the cURL client, what is the response? It seems like your service wasn't created properly. Is there any relevant information in the syslogs? The mentioned error occurs when establishing a parent-child relationship internally within the service API endpoint.

Dibyajyoti-08, 2024-12-17:
With the gui client or the curl client i am getting the same response as **no such category exist.**
This is the syslog after trying to send the curl request to register the plugins,
```bash
[1987] ERROR: logger: fledge.common.configuration_manager: Unable to create new category based on category_name zigbee_uart and category_description Zigbee UART Data Communication Module and category_json_schema

[1987] WARNING: common: fledge.services.core.api.plugins.common: Python plugin not found......module 'fledge.plugins.south.zigbee_uart.zigbee_uart' has no attribute 'plugin_info', try C-plugin
```

Dibyajyoti-08, 2024-12-17:
[image]
[image]

ashish-jabble, 2024-12-17:
> [1987] ERROR: logger: fledge.common.configuration_manager: Unable to create new category based on category_name zigbee_uart and category_description Zigbee UART Data Communication Module and category_json_schema

Category creation failed, which indicates that the JSON schema is incorrect. Based on the information from your plugin, I see that the default value for the configuration item **baudrate** is 9600, which is an integer. However, in the Fledge configuration, all values should be passed as strings, regardless of their defined type, therefore "9600". That said, there are some optional attributes in the configuration that do not necessarily need to be passed as strings.
See the configuration life cycle [here](https://fledge-iot.readthedocs.io/en/develop/plugin_developers_guide/02_writing_plugins.html#configuration-lifecycle)

Dibyajyoti-08, 2024-12-17:
is this the reason its trowing the error of "no such category exist"?

closed 2024-12-17 by ashish-jabble