fledge-iot/fledge #1516: North plugin shutdown not triggered during configuration changes

kwitekrac, 2024-12-13:
I encountered an issue while testing a simple MQTT South plugin with an IEC104 North plugin in Fledge 2.5.0. The issue appears to originate from Fledge Core or [dispatcher](https://github.com/fledge-iot/fledge-service-dispatcher) and is not plugin-specific. Below are the details of the scenarios tested and the observations:

*Scenario 1:*
1. Start Fledge 2.5.0.
2. Set the log level to DEBUG and enable the North plugin via the UI.
3. Configure the North plugin, exchanged_data, and protocol_stack filters using the REST API.
4. Send data to the South plugin for the first time—this creates a new asset intended for forwarding to the North plugin.
5. Wait 1 minute and trigger a data change.

After the configuration changes via the REST API, the `plugin_shutdown` function for the North plugin was not triggered. It was only triggered after the South plugin received its first set of data.

*Scenario 2:*
1. Start Fledge 2.5.0.
2. Send data to the South plugin for the first time—this creates a new asset.
3. Set the log level to DEBUG and enable the North plugin via the UI.
4. Configure the North plugin, exchanged_data, and protocol_stack filters using the REST API.
5. Wait 1 minute and trigger a data change.

In this scenario, after step 4 (configuration change), the `plugin_shutdown` function was triggered, and the data received in step 5 was forwarded correctly.

This issue was discussed and described in detail in [IEC104 North Plugin repo](https://github.com/fledge-power/fledge-north-iec104/issues/57)

MarkRiddoch, 2025-01-24:
The calling of plugin_shutdown when reconfiguring is to overcome the lack of  a plugin_reconfigure int he north plugins and is an historical artefact dating from when we only had tasks in the north and not services. The service will call plugin_reconfigure if it detects the north plugin has one, otherwise it will call shutdown and create a new instance of the plugin, calling plugin_init again. Do you by an chance have a plugin_reconfigure method defined for the IEC 104 North plugin?

kwitekrac, 2025-01-27:
No, I'm using the [Fledge Power IEC104 North Plugin](https://github.com/fledge-power/fledge-north-iec104/blob/main/src/plugin.cpp), and indeed, `plugin_reconfigure` is not implemented. However, `plugin_shutdown` followed by `plugin_init` should be sufficient to apply configuration changes.

The open question is why these functions are only triggered when data has been previously received (Scenario 2) but are not triggered if no data was transmitted prior to the configuration change (Scenario 1).

MarkRiddoch, 2025-02-07:
I can't see any obvious reason for this behaviour. The only thing I can se that might cause an issue is that we block the reconfigure if there is a plugin_send operation in progress and only do the shutdown/restart after that plugin_send completes. Is it possible the IEC104 north is blocking inside plugin_send?

kwitekrac, 2025-02-12:
I don't think so, as in the testing environment, I ingest a single data sample by running the script—so I'm sure `plugin_send` was not triggered. Without knowing the Fledge core code, I would say that triggering `plugin_send` initializes something that is required for reconfiguring the plugin.

FlorentP42, 2025-02-12:
> The calling of plugin_shutdown when reconfiguring is to overcome the lack of a plugin_reconfigure int he north plugins and is an historical artefact dating from when we only had tasks in the north [...]

This is an interesting information, does this mean that plugin_reconfigure is now supported on north plugins? Since which Fledge version? It does not appear to be documented anywhere : https://fledge-iot.readthedocs.io/en/latest/plugin_developers_guide/04_north_plugins.html