fledge-iot/fledge #1523: Small clarification on custom plugins

Dibyajyoti-08, 2024-12-20:
Hi Folks,
Recently i added a custom plugin in the Fledge and did some testing, i stumbled across one doubt, that the plugin that i have added to my south service is not enabled, the status is nothing, 
Can you suggest any document to take reference or the way to work it out.
[image]
[image]

ashish-jabble, 2024-12-23:
@Dibyajyoti-08 It seems like this issue might be related to service registration. Are there any relevant syslogs available?

You can follow the [guide for testing a new plugin](https://fledge-iot.readthedocs.io/en/latest/plugin_developers_guide/10_testing.html) for assistance.

For further clarification, feel free to join the `#fledge` or `#fledge-help` channels on [Slack](slack.lfedge.org) instead of creating issues.

Additionally, could you please share your use case with Fledge and specify which organization you're associated with? This will help us address your challenge more effectively.

Dibyajyoti-08, 2024-12-23:
sure @ashish-jabble i will mail you the details of the use cases as well the the association details

MarkRiddoch, 2025-01-24:
The must likely cause of this is if there is an issue in the default configuration for the plugin. It fails to load and should write an error to the error log.