fledge-iot/fledge #1526: south mqtt_reading plugin shutdown

shreyadanole, 2024-12-25:
I encountered an issue while testing a simple MQTT South plugin in Fledge 2.6.0. The issue seems to originate from Fledge Core. Below are the details of the scenarios tested and the observations:

Scenario 1:

Start Fledge 2.6.0.
Configure the South plugin for mqtt_reading.
The South service mqtt shuts down automatically.
Each time, we need to reinstall the mqtt plugin and restart Fledge.
[image]

ashish-jabble, 2025-01-02:
Hi @shreyadanole Thanks for showing your interest in Fledge. 
Could you please clarify the following points?
A) Is the Fledge set up with source code, a package, or Docker-based images?
B) What is the environment platform you're using?
C) Are you using the latest version of the Fledge, or a nightly/development build?
D) There's no apparent reason for the abrupt shutdown. Would you mind sharing your configuration information? It would be helpful if you could provide the support bundle of your setup.

shreyadanole, 2025-01-03:
Hi @ashish-jabble, I added the following points related to Fledge:

A) The Fledge is set up using Docker-based images.
B) The environment platform I using for Fledge is Windows 10.
C) The version of Fledge I using is 2.6.0.
[image]
[image]
[image]

ashish-jabble, 2025-01-03:
@shreyadanole As per the first screenshot, the logs indicate a "Module Not Found" error, which suggests that the `paho-mqtt` package might not have been installed correctly. Could you please check the installed packages by running the `python3 -m pip freeze` command?

If you are using the latest version (v2.6.0), then the correct version of paho-mqtt should be **paho-mqtt==1.4.0**. However, I’m confused about how you were able to add its service, considering the log explicitly states that the module was not found. Did you manage to fix it on your own?

Also, assuming that the Mosquitto host and port are reachable from the Docker container.

You can also enable debug logging by setting the **Minimum Log Level** option inside the _Advanced_ tab. By default, this is set to warning.

Additionally, please share the support bundle, which can be generated via the GUI client. You can do this by navigating to `Support -> Request New -> Download`. Once downloaded, please share the zip folder.

shreyadanole, 2025-01-06:
Hi @ashish-jabble, I have set up Fledge using Docker-based images. The first time the MQTT reading South plugin was available, I did the configuration, and it worked fine.
However, after I shut down the containers and spin them up again, I am facing the following issue. 
[image]

I have added the support bundle below.
[support-250106-07-09-18.tar.gz](https://github.com/user-attachments/files/18314932/support-250106-07-09-18.tar.gz)

ashish-jabble, 2025-01-06:
Hi @shreyadanole Thank you for providing the support bundle. I reviewed the syslogs, and as mentioned earlier, they indicate a 'ModuleNotFoundError'. See the logs:
```
Jan  6 07:08:08 025e766e2fcf Fledge DEMO-MQTT[687]: WARNING: logErrorMessage: Error 'ModuleNotFoundError("No module named 'paho'")', plugin 'mqtt-readings'
Jan  6 07:08:08 025e766e2fcf Fledge DEMO-MQTT[687]: WARNING: Traceback (most recent call last):#012#012  File "/usr/local/fledge/python/fledge/plugins/south/mqtt-readings/mqtt-readings.py", line 51, in <module>#012    import paho.mqtt.client as mqtt#012#012ModuleNotFoundError: No module named 'paho'
Jan  6 07:08:08 025e766e2fcf Fledge DEMO-MQTT[687]: FATAL: PluginInterfaceInit: cannot import Python 3.5 script 'mqtt-readings' from '/usr/local/fledge/python/fledge/plugins/south/mqtt-readings' : plugin 'mqtt-readings'
Jan  6 07:08:08 025e766e2fcf Fledge DEMO-MQTT[687]: ERROR: Python plugin mqtt-readings does not support plugin_info entry point.
Jan  6 07:08:08 025e766e2fcf Fledge DEMO-MQTT[687]: FATAL: Failed to load south plugin DEMO-MQTT, exiting...
```

Upon checking the installed software and Python pip packages, I noticed that both the `mqtt-readings` software and the `paho-mqtt` pip dependency are missing from the support bundle. This suggests that the environment is lacking the necessary MQTT-related components, which is likely causing the service to shut down. I’m still investigating the issue, and as you mentioned, the problem persists after restarting the containers ONLY.

How are you shutting down or restarting the containers? Are external volumes involved in the container setup? If so, this might be affecting the dependencies or configuration. Otherwise, there shouldn’t be any obvious reason for this issue to occur.

Also, could you clarify when the support bundle was created—was it generated during the initial startup or after the containers were shut down and you observed the service shutting down completely?

shreyadanole, 2025-01-06:
I am shutting down the containers using Docker commands. There are no external volumes involved. The support bundle was created after the containers were shut down and then brought back up again.

ashish-jabble, 2025-01-06:
> when the containers are down without volume options the mqtt-readiings plugin is lost and we need reinstall the plugin unlike in version 1.9.2

I'm not sure what you mean by "when the containers are down without volume options." Could you clarify?
Also, the official Docker images are available starting from version 2.0 onwards. Not sure about your 1.9.2 version

I followed the setup instructions using the [Docker image](https://fledge-iot.readthedocs.io/en/latest/quick_start/installing.html#ubuntu-20-04), installed mqtt-readings, created the service, and ingested readings using the [existing script](https://github.com/fledge-iot/fledge-south-mqtt/blob/v2.6.0/mqtt-pub/__main__.py) from the plugin repository. After some time, I stopped the container with `docker stop <container_name>` and then started it again with `docker start <container_name>`. I did not encounter any errors, and the service remained running, with readings being ingested again as expected.

For further clarification, feel free to join the `#fledge` or `#fledge-help` channels on [Slack](slack.lfedge.org) or ping me there

ashish-jabble, 2025-01-17:
There’s no issue on our end, but the commands used were incorrect. I’ve provided comments in the 'fledge-help' Slack channel for clarification

> when I stop the image using the command docker compose down and then spin it up again using docker compose up -d, I face an issue where the installed plugins automatically shut down.

This command stops and removes containers created for the fledge service, and also removes any associated networks. However, it does not remove volumes by default. If the expectation is that the plugin installation will be preserved, this will not be the case

**Solution:** Use `docker-compose stop`, which does the following:

_Stops running containers:_ The containers for the services are gracefully stopped but not deleted.
_Keeps volumes and networks:_ Volumes and networks are kept intact. The data stored in volumes (if any) remains safe.
_Containers remain in “stopped” state:_ The containers remain on your system, but they are not running. You can restart them later with `docker-compose start` or `docker-compose up`.

Therefore, the issue is considered resolved, as agreed upon with the reporter.