fledge-iot/fledge #1527: Missing logs in fledge version 2.6.0 for south service

skumarorxa, 2024-12-27:
No logs shown for in latest fledge version 2.6.0 for example if we create mqtt reading service. The same service shows logs is the older version 1..9.2. Please fix at the earliest

MarkRiddoch, 2024-12-27:
Are there particular logs you are looking for? We made changes to make the logging less verbose by default, you may need to turn up the logging level to get the logs of things that previously, and incorrectly, appeared by default.

ashish-jabble, 2025-01-17:
Here is the [documentation link](https://fledge-iot.readthedocs.io/en/latest/tuning_fledge.html#south-service-advanced-configuration) for configuring log settings under the Advanced tab of a South service. Similarly, the same process applies for configuring logs under the Advanced tab of a North service.