fledge-iot/fledge #1528: Using using docker container when is down the institution is removed

skumarorxa, 2024-12-27:
When using fledge running as docker container using docker-compose image from robraesemann/fledge:2.6.0. When the container is down the installed south package say mqtt-readings disappears in the GUI and logs throw up error 
South service MSEDCL-STMS1-JSON is shutting down due to a failure loading the south plugin

ashish-jabble, 2025-01-02:
Hi @skumarorxa Thank you for showing interest in the Fledge. You can find the official Docker images [here](https://fledge-iot.readthedocs.io/en/latest/quick_start/installing.html#using-docker-containerizer-to-install-fledge) and try running your setup with them.

FelixHelin, 2025-01-04:
@ashish-jabble Why a private registry and not Dockerhub? Is the Dockerfile used to build the images also made public?

ashish-jabble, 2025-01-17:
> Why a private registry and not Dockerhub? Is the Dockerfile used to build the images also made public?

Currently, the Docker images are hosted on a private registry, but they will be available on Docker Hub in future releases.

Issue relates with #1526 . See [comments](https://github.com/fledge-iot/fledge/issues/1526#issuecomment-2597994180)

closed 2025-01-17 by ashish-jabble