fledge-iot/fledge #1546: Fledge ENV vraibles

skumarorxa, 2025-02-03:
Which ENV variables are used to enable authentication is fledge say "basic" authentication when fledge container runs.

MarkRiddoch, 2025-02-07:
Authentication within Fledge is configured via the configuration category. The only environment variables used are to set the route of the installation if it is not installed in the default location. A few debugging options are set via the environment, but they are not for production use.

You can also set the PostgreSQL related environment variables to point Fledge at your PostgreSQL installation if it is outside of the same machine, VM or container.

MarkRiddoch, 2025-02-20:
The next release of Fledge is going to turn on authentication by default for all new installations. For existing implementation take a look at the documentation. section [Securing Fledge](https://fledge-iot.readthedocs.io/en/latest/securing_fledge.html)