fledge-iot/fledge #1551: As of now you can insert dictionary key pair { "key": value } only in the readings

skumarorxa, 2025-02-10:
readings has the following attributes asset_code, timestamp, reading etc. I want to insert another attributes "events" which is filter using python35 the events is list of dictionary. In the format "events" : [ { "name": "CB", "value": 1 } , {"name": EF : "value": 0 }] but able to insert this key asset in filter readings when using python35. The filter is working without error but in the Http north plugin when posting this data to the DB I do not see this "event" in payload. As of payload can accept simple Key value pair only. How to implement this use case

MarkRiddoch, 2025-02-20:
Are you inserting the events below the readings, i.e. as a datapoint in the readings or alongside the readings? If it is the later, then no, you can not do this as the readings JSON format has to have a standard layout. Maybe you could post an example of the JSON structure you are creating so that it is clearer to us.

skumarorxa, 2025-02-21:
Thanks for reply I want to add data point below the readings in this format "readings": { "Asset": "TEST01", "timestamp": "2025-02-21 08:00:00", "reading": { "DP1":0, "DP2": 1: "DP3": 2 }, "events":  : [ { "name": "CB", "value": 1 } , {"name": EF : "value": 0 }]